ttgapers/cakeqos-merlin #2: Additional Parameters not accepting more than one

ttgapers, 2020-06-19:
Currently any command with Additional Parameters set to more than one throws verbose error from cake and only sets the first parameter on the command line in the start/stop scripts.

/jffs/scripts/cake-qos enable 135Mbit 13Mbit "docsis ack-filter"

Looks like we aren't splitting the passed multiple parameters correctly.

jkychn, 2020-06-19:
To test (more verbose instructions below if anyone else is following along and wants to test):

`/usr/sbin/curl --retry 3 "https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/develop/cake-qos-firewall-nodelay.sh" -o "/jffs/scripts/cake-qos" && chmod 0755 /jffs/scripts/cake-qos`

Skip the below command if you already have the cake binaries installed:
If ac86u: `/jffs/scripts/cake-qos install ac86u`
If ax88u: `/jffs/scripts/cake-qos install ax88u`

`/jffs/scripts/cake-qos enable 135Mbit 13Mbit "docsis ack-filter"`

Edit the **135Mbit 13Mbit "docsis ack-filter"** portion in the command above to suit your needs.

Then reboot and verify afterwards with:

```
tc qdisc
tc -s qdisc show dev eth0 (for upload)
tc -s qdisc show dev ifb9eth0 (for download)
```

And/or a test on dslreports.com/speedtest

jkychn, 2020-06-19:
Confirmed fixed in v0.0.2 by TechTinkerer!

closed 2020-06-19 by jkychn