ttgapers/cakeqos-merlin #3: Autodetect routers based on Martineau's script

ttgapers, 2020-06-21:
See: https://www.snbforums.com/threads/release-cakeqos-merlin.64800/page-5#post-595470

`Get_Router_Model() {
    local HARDWARE_MODEL
    [ -z "$(nvram get odmpid)" ] && HARDWARE_MODEL=$(nvram get productid) || HARDWARE_MODEL=$(nvram get odmpid)
    echo $HARDWARE_MODEL
    return 0
}
Is_HND() {
    [ -n "$(uname -m | grep "aarch64")" ] && { echo Y; return 0; } || { echo N; return 1; }
}
Is_AX() {
    # Kernel is '4.1.52+' (i.e. isn't '2.6.36*') and it isn't HND
    [ -n "$(uname -r | grep "^4")" ] && [ -z "$(uname -m | grep "aarch64")" ] && { echo Y; return 0; } || { echo N; return 1; }
}

### Check Requirements

FAIL="0"
if [ "$(nvram get jffs2_scripts)" -ne 1 ]; then
    echo "ERROR: Custom JFFS Scripts must be enabled."
    FAIL="1"
fi
if [ ! -f "/opt/bin/opkg" ]; then
    echo "ERROR: Entware must be installed."
    FAIL="1"
fi
#################### Martineau Hack #############################
HARDWARE_MODEL="$(Get_Router_Model)"
if [ "$(Is_HND)" == "N" ] && [ "$(Is_AX)" == "N" ];then
    echo -e "\a\t[✖] ***ERROR Router $HARDWARE_MODEL isn't supported (Only HND-models RT-AC86U,RT-AX88U or RT-AX56U,RT-AX58U)"
    FAIL=1
fi
#################################################################
if [ "${FAIL}" = "1" ]; then
    return 1
fi`

@jkychn thoughts on incorporating?

maghuro, 2020-06-22:
#8

ttgapers, 2020-06-22:
> 
> 
> #8

Awesome @maghuro - pull request to be reviewed as well!

closed 2020-06-22 by maghuro