ttgapers/cakeqos-merlin #11: Automatically calculate 95% download/upload speeds

jkychn, 2020-06-22:
A suggestion from TechTinkerer:

```
Could bit of code be used to calculate 85-90% as part of the cake-qos script?

I had this in freshjr_qos could it be adapted, so you just input speedtest result and it outputs speeds on a percentage and loads it into the cake-qos

Code:
DownCeil="$(printf "%.0f" $(nvram get qos_ibw))"
UpCeil="$(printf "%.0f" $(nvram get qos_obw))"
DownCeil="$(expr ${DownCeil} \* 95 / 100)"
UpCeil="$(expr ${UpCeil} \* 95 / 100)"

DownCeil="$(printf "%.0f" $(nvram get qos_ibw))"
UpCeil="$(printf "%.0f" $(nvram get qos_obw))"
DownCeil="$(expr ${DownCeil} \* 95 / 100)"
UpCeil="$(expr ${UpCeil} \* 95 / 100)"
```

Creating this issue for discussion.

ttgapers, 2020-06-22:
Interesting. Assuming all the pre-reqs are met, this is not a bad idea for "set and forget" type users. @jkychn @maghuro thoughts around this as a medium term addition as we stabilize the release?

maghuro, 2020-06-22:
I don't agree.
In think people have to choose what they want.

This way we are forcing always 95% of what people choose.

Let's say I want 90% from my actual speed?

Everything that is forced must not be in. Just my two cents :)

ttgapers, 2020-06-22:
Fair enough @maghuro as it also goes inline with if you don't enter any up/down it defaults to line speed. I am on your page for sure. In addition one bad run can potentially have adverse effects.

@jkychn - thoughts?

jkychn, 2020-06-22:
Agreed, thanks :) Closing!

closed 2020-06-22 by jkychn