ttgapers/cakeqos-merlin #12: "update" option md5 issue

jkychn, 2020-06-22:
```
KBvUTgSItSQdjpCB@RT-AC86U-87E8:/tmp/home/root# /jffs/scripts/cake-qos update ac86u
/jffs/scripts/cake-qos: line 331: #!/bin/sh: not found
CakeQOS-Merlin v0.0.5: Local and server md5 don't match, updating...

CakeQOS-Merlin v0.0.5: Your cake binaries are up-to-date.

KBvUTgSItSQdjpCB@RT-AC86U-87E8:/tmp/home/root# /jffs/scripts/cake-qos update ac86u
/jffs/scripts/cake-qos: line 331: #!/bin/sh: not found
CakeQOS-Merlin v0.0.5: Local and server md5 don't match, updating...

CakeQOS-Merlin v0.0.5: Your cake binaries are up-to-date.
```

jkychn, 2020-06-22:
@maghuro if you could please look into this. Thank you :)

jkychn, 2020-06-22:
Just received an update: `It fixed itself when i ran update check again`

If you could confirm this isn't an issue feel free to close this. Thanks!

maghuro, 2020-06-22:
Maybe it's because you're putting the model in front. Don't need it 

I can only go to my PC and debug everything I've commited this night

maghuro, 2020-06-22:
@ttgapers can you please give me write access? so I can pull directly to develop?

ttgapers, 2020-06-22:
> 
> 
> @ttgapers can you please give me write access? so I can pull directly to develop?

Done I thought I did it already. Let me know if you have issues @maghuro

maghuro, 2020-06-22:
> > @ttgapers can you please give me write access? so I can pull directly to develop?
> 
> Done I thought I did it already. Let me know if you have issues @maghuro

I don't have write access yet...

maghuro, 2020-06-22:
oh i had to accept the invite! :) :P

maghuro, 2020-06-22:
Fixed: https://github.com/ttgapers/cakeqos-merlin/commit/f6b7ed8c7b95a5baaad5298b4bee97bf5089513e

closed 2020-06-22 by maghuro