ttgapers/cakeqos-merlin #31: Start if line present

maghuro, 2020-06-23:
@JackYaz
I need your expertise.

Everytime we use option "stop", the cake stops and the start line on nat-start keeps untouched. That's the correct way!

However, when pressing start, we have to introduce the values again.

My suggestion is, if possible, change the option "stop" to "toggle"

When opening the menu, the menu would check if cake is running or not and stick this info above the menu. Then the toggle button would do it's job (stop if running and start if stopped).

However, if one presses toggle and the start line is on the nat-start, it'll use that line with that params to start cake.
If it isn't present on nat-start, then the script would ask for all the parameters needed.

Note that the option 1 (start) should always ask for parameters. In fact it could be renamed to enabled/configure or whatsoever.

Did I made me understand? 😊

Thanks.

jackyaz, 2020-06-23:
Sure, I can do that but it will be sometime tomorrow

ttgapers, 2020-06-23:
@jackyaz thanks for all your help on this thus far! If it's easier I can make you send you a contributor invite...

ttgapers, 2020-06-23:
@jackyaz thanks for helping out on this project!

Adamm00, 2020-06-24:
This has been addressed in a future update.

maghuro, 2020-06-24:
Fixed.

closed 2020-06-24 by maghuro