ttgapers/cakeqos-merlin #33: cake-qos start not changing nat-start

ttgapers, 2020-06-24:
Let's see if I am doing something wrong, but looks like cake-qos start up down "additional parameters" doesn't change diffserv4 to besteffort etc.

If I uninstall and issue the start command after pulling the script down again, all is well. I'll have a look at the code, but I know I am playing catch up.

ttgapers, 2020-06-24:
Looks like update should do it..will verify and close.

jkychn, 2020-06-24:
Probably related to this line: https://github.com/ttgapers/cakeqos-merlin/blob/master/cake-qos.sh#L202

What does running `grep -qF "# CakeQOS-Merlin" /jffs/scripts/nat-start` in terminal return?

Adamm00, 2020-06-24:
I have addressed this (and other issues) in my local fork, so hold tight until its ready to go live.

maghuro, 2020-06-24:
Can you check it's working now with #34 ?

maghuro, 2020-06-24:
Tested. Fixed

closed 2020-06-24 by maghuro