ttgapers/cakeqos-merlin #37: Menus

maghuro, 2020-06-24:
When using option 4 to set speeds and params, when the value is inserted the break makes it to go to main menu, instead of going to the "Edit configs" menu.
Is it possible to make it go one menu back and not two menus back?

Adamm00, 2020-06-25:
Not possible with how we pass on the values from selecting menu options. Every time after you select a menu option and it executes the corresponding case statement function, we actually start a new instance of the script.

maghuro, 2020-06-25:
Can you please test if this commit solves this "issue"? 
I'm not at home right now @ #

Adamm00, 2020-06-25:
No it doesn't, like I said, the way the menu is designed this functionality isn't possible. In future please test commits locally before committing, keeping an organized commit log is very important in a open source project.

closed 2020-06-25 by Adamm00