ttgapers/cakeqos-merlin #44: After update go back to Cake-QOS Menu

ttgapers, 2020-06-27:
Not a high priority, but I am not sure if after running Cake update, it takes you back to the Menu that out to the shell prompt. 

If it's not possible please close. If it is, we can track it here.

@Adamm00 your thoughts appreciated.

Adamm00, 2020-06-28:
Not possible. If you overwrite the script (aka update it) you have to immediately exit otherwise the shell will get confused on what line to execute and error out.

jackyaz, 2020-06-28:
Can you not call "exec $0" and then exit the previous instance. That's how unbound_manager and my scripts handle reloads

Adamm00, 2020-06-28:
I can't seem to reproduce the issue which was present in commits prior to 38526b3493856a54e0e8313c17e2b277f35eb3d5 so I assume there are more specific circumstances that trigger this behavior (the same which I have previously experience in Skynet). With that being said, I've gone ahead and removed the hard exit in 7752ddcb4225953f5943d8952ecf3642879e1303 as it _appears_ to be working correctly for now.

closed 2020-06-28 by Adamm00