ttgapers/cakeqos-merlin #46: Allow Extra Options for upload and download individually

BrummyGit, 2020-06-28:
As requested by a few on the forum, please consider allowing extra options to be specified separately for Upload and Download. There are some users who would like to fix the bandwidth in one direction (usually download) but have variable bandwidth in the other (usually upload). Also some of the Flow Isolations parameters are more useful in one direction compared to the other.

This could be a standard set, then additions for upload / download only or any other implementation.

Thanks for considering

Adamm00, 2020-06-28:
As per my comment in the thread, is there a real world example you can list where this type of functionality would be useful? No one so far has been able to list a compelling argument for this to be added.

RobotsAreCrazy, 2020-06-28:
@Adamm00 it's just so i don't have to go manually editing the script to add extra options that i would want to be different per upload\download

Adamm00, 2020-06-28:
> @Adamm00 it's just so i don't have to go manually editing the script to add extra options that i would want to be different per upload\download...not bothered about connection profiles the main thing is being able to have different options per up\down

Do you have any real world examples? My point being this change would add an extra layer of complexity to new users and no one so far has been able to list an actual example of settings that would differ in a typical use-case.

BrummyGit, 2020-06-28:
Use cases seem to focus mainly around the use of _autorate-ingress_ and _unlimited_ bandwidth settings. They tend to focus on asymetric bandwidth connections which are sometimes variable in one direction. It has also been suggested that people may wish to:

1. Use unlimited on fast a download connection which is not problematic while applying different parameters to a slower, more congested upload
2. Use Automatic rate adjustment on the upload which varies through the day, whilst using a fixed bandwidth on the download which remains consistent

For example in the forum maghuro says:

> In my particular case, which have an LTE 40/10 connection, it'd be useful.
> For defining a fixed bandwidth on download, but to have for example autorate-ingress on upload, which is very floating. My upload floats very much between 1 and 10 mbit.
> As for download, it's always on 35 more or less.

There are other requests which are similar.

You could offer _unlimited_ on just one direction easily by allowing 0 as valid for a bandwidth setting, but _autorate_ingress_ is more complex to set in one direction at a time as it can be used in conjunction with an initial bandwidth setting. To set without an initial bandwidth you could also catch "A" or "Auto" in the bandwidth field to set _autorate_ingress_ in a single direction instead of a bandwidth value.

I've also seen discussion about using _dual-srchost_ and _dual-dsthost_ flow isolation parameters which are more suitable for one direction rather than the other. Reading the _triple-isolate_ description it seems that this is the best of both options working in both directions so maybe this use case is not as strong.

BrummyGit, 2020-06-28:
Just to add that in the last couple of hours I've been testing the use of unlimited download bandwidth (set using 0 as my bandwidth in the config file) and setting a fixed (95%) value for my upload bandwidth. My bufferbloat remains controlled by just shaping my upload where the bottleneck is for my router, giving maximum burst speeds on my download.

If others have the same experience, the ability to set 0 in each of the bandwidth settings would be very useful and should be simple to implement, but I recognise that it doesn't solve the request for additional parameters to be set as made by others.

Adamm00, 2020-06-29:
> I've also seen discussion about using dual-srchost and dual-dsthost flow isolation parameters which are more suitable for one direction rather than the other. Reading the triple-isolate description [...]

**triple-isolate** is enabled by default, I don't see any convincing reason people would want to downgrade their experience.

As for **autorate_ingress**, am I wrong in thinking that lets say you had a stable download but unstable upload, when this setting is applied in both directions it would only have a noticeable effect on the upload and the download rate would stay steady?

BrummyGit, 2020-06-29:
I have to agree regarding **triple-isolate** - anyone desperate enough to use it would in theory be capable of editing their own scripts.

For **autorate-ingress** that's not correct. Both directions fall back towards zero and then increase over a few seconds on demand. I ran many speed tests using a number of different tools and it's always the same. That's why I was saying people would want to set their download to a fixed (and therefore stable) value but use **autorate-ingress** on their upload which doesn't provide consistent performance. This is most useful in asymmetric services.

Lastly I'm wondering whether to open a separate issue for consideration on allowing **0** as a valid bandwidth value in the field validation. It's supported officially by cake and allows **unlimited** to be set in a single direction. Again this is most useful for asymmetric configurations where people have plenty of bandwidth available for their download (often over the cake recommended maximum) but limited upload bandwidth which suffers. In this case they can remove the shaping of the download (which I expect would improve the throughput) whilst maintaining good bufferbloat on the upload. Is it worth breaking this out separately?

Adamm00, 2020-06-29:
I've pushed the changes to the **v1.0.2** branch, this should seamlessly transition your old unified settings to the two new individual settings and allow you to configure them. Let me know if there are any issues.

ttgapers, 2020-07-01:
Now getting a chance to test out the new builds. Thanks @Adamm00

ttgapers, 2020-07-01:
Updated and was able to add good and got errors on bad settings as well. Seems ok, and other issues reported? I can play some more with the up/down settings as well.

Adamm00, 2020-07-01:
Added in #52

closed 2020-07-01 by Adamm00