ttgapers/cakeqos-merlin #50: Cater for github being down

BrummyGit, 2020-06-29:
I've updated my script this morning following the change to allow 0 bandwidth and the script keeps freezing. I think there might be problems with github.com as my web browser access is also intermittent today.

I think the issues with the cake-qos script freezing is that it performs an update check every time it hits the main menu. I had noticed the script has slowed down and I think it's the update checks. Can we consider 2 related changes:

1. Add a timeout to the update check in case github is down
2. Only auto-check for updates once per launch

Adamm00, 2020-06-29:
> Add a timeout to the update check in case github is down

This is already the case, we take 3 attempts at downloading the manifest each attempt can only a maximum of 3 seconds meaning in the worst case scenario the menu will hang for 9 seconds before completely timing out.

> Only auto-check for updates once per launch

This wouldn't really have any effect, every time you choose a menu option and it successfully executes a new instance of the script is launched. So even if we add a one time check the only case this will apply is when you go into then back out of the settings menu.

Fortunately github outages are quite infrequent (and so should people modifying their settings) so this is a compromise that we have to make.

closed 2020-06-29 by Adamm00

jackyaz, 2020-06-29:
This is caused by the design of the menu system. Other designs don't have to make this compromise.

Adamm00, 2020-06-29:
Good joke

Adamm00, 2020-06-29:
To elaborate, this is a design choice, not flaw. We intentionally execute a new instance of the script rather then running the Cake_Menu function again.

https://github.com/ttgapers/cakeqos-merlin/blob/7e14b6a5befd4add969895462137c9f39bec9515/cake-qos.sh#L668

We do this for numerous reasons, the one of them being so we can reload with a clean slate, including reloading any settings that may have been changed manually by the user or another instance of the script. This gives us significantly more flexibility as the menu options become more complex as in other scripts where it is used.