ttgapers/cakeqos-merlin #51: Additional Menu Items > Status

ttgapers, 2020-07-01:
@Adamm00 @jackyaz 
Not sure how how difficult it would be to get the following added to perhaps an Option 4 sub-menu:

4a: Upload Stats: tc -s qdisc show dev eth0 # for upload
4b: Download Stats: tc -s qdisc show dev ifb9eth0
4c: tc qdisc show

Thoughts?

Adamm00, 2020-07-01:
d6eee17c07564d4ed1dbfca86ee6a6468d0884c6

ttgapers, 2020-07-01:
Will update - thanks!

Adamm00, 2020-07-01:
> Would be good to go back to Option 3 menu than all the way out

Not sure what you mean by this

> Is this one a bit more difficult than the other 2? Show traffic control settings: tc qdisc show

We show the important parts of this under the general option.

Adamm00, 2020-07-01:
Added in #52

closed 2020-07-01 by Adamm00

ttgapers, 2020-07-01:
> 
> 
> > Would be good to go back to Option 3 menu than all the way out
> 
> Not sure what you mean by this
> 
> > Is this one a bit more difficult than the other 2? Show traffic control settings: tc qdisc show
> 
> We show the important parts of this under the general option.

First item ugandy mentions here as well: https://www.snbforums.com/threads/release-cakeqos-merlin.64800/page-40#post-599088

Steps from Main Menu:
Select 3 (Cake Status)
Then pick any Status option
Hit any key to continue -- takes you back to the Main Menu, and not the Status Menu (preferred)

Hope that helps.

Adamm00, 2020-07-01:
That's unfortunately not how the menu system works, it would require significantly messier code (every menu/submenu would need to be a function) and add unnecessary complexity for minimal benefit.

While it may be exciting to constantly tinker with cake because its a shiny new toy, it is designed to be a set and forget solution. Interaction with the script at all should be almost non-existent post configuration in a regular use-case outside debugging.

ttgapers, 2020-07-01:
I agree...I've settled on the defaults as my "best" for everyone. besteffort docsis ack-filter

Just thought it would be easy sub-menu-ing if just checking status and queues/tins. Just and extra couple inputs to get back to the Sub-menu screen. If it's too much effort, then kindly park it :) I can live with the extra clicks!

I keep pushing the KISS mantra -- don't ever worry. It's what started me here. Flex and FreshJR has too many configs to get it to a stable/working state. This just works.