ttgapers/cakeqos-merlin #54: Command "cake-qos status" not recognized

pro-sumer, 2020-07-02:
```
> cake-qos help

CakeQOS-Merlin: start:     start cake-qos
CakeQOS-Merlin: stop:      stop cake-qos
CakeQOS-Merlin: status:    check the current status of cake-qos
CakeQOS-Merlin: settings:  configure settings for cake-qos
CakeQOS-Merlin: update:    update cake-qos & binaries (if available)
CakeQOS-Merlin: install:   install and configure cake-qos
CakeQOS-Merlin: uninstall: stop cake-qos, remove from startup, and remove cake binaries

> cake-qos status

Command Not Recognized, Please Try Again
```

Please support the "status" command (or update the help text).

pro-sumer, 2020-07-02:
Same for "settings".

Adamm00, 2020-07-02:
Additional arguments are required;

`cake-qos status download
cake-qos status upload
cake-qos status general

cake-qos settings dlspeed xxx
cake-qos settings upspeed xxx
cake-qos settings queueprio {besteffort|diffserv3|diffserv4|diffserv8}
cake-qos settings optionsdl xxxx
cake-qos settings optionsul xxxx
`

This script will also display the CLI command equivalent when you execute a menu option

pro-sumer, 2020-07-02:
Had no idea after reading the help.

May display this?

`CakeQOS-Merlin: status [download | upload | general]: ...`

Adamm00, 2020-07-03:
Fixed in fb61052244520c33ada4ca5bf29dc2cc4bab0cf5

closed 2020-07-03 by Adamm00