ttgapers/cakeqos-merlin #57: cakeQoS doesn't survive a Trend Micro signature update check

BrummyGit, 2020-07-16:
cakeQoS doesn't survive a Trend Micro signature update check if I run it manually from the gui. From something I read on the FlexQoS thread, this will also be an issue in the event that an automatic signature update is applied.

Router log shows:

> Jul 16 10:30:38 rc_service: httpd 1257:notify_rc start_sig_check
> Jul 16 10:30:38 custom_script: Running /jffs/scripts/service-event (args: start sig_check)
> Jul 16 10:30:42 kernel: The For ALL DEVICES flag of Prof 1 has been set to ENABLE
> Jul 16 10:30:42 BWDPI: fun bitmap = 47f

Adamm00, 2020-07-18:
I can't reproduce this on the latest commit (384.19_alpha1-gd167a9267) on my AX88U. So this is either device specific or there are other factors involved which cause this (I also have AiProtect enabled fwiw). 

In any case, the watchdog _should_ automatically fix this during its hourly checks.

BrummyGit, 2020-07-18:
OK fair enough - FYI also running an AX88U but with the later 384.19_alpha1-g2546f95291 firmware build. I hadn't realised that the hourly watchdog would pick this up - will test this.

BrummyGit, 2020-07-19:
I tested and as expected the Trend Micro signature check killed cake-qos but the hourly watchdog check restarted it so worst case is cake-qos not running for 1 hour.

ttgapers, 2020-07-20:
Hi @Adamm00 

Just getting to follow up on this one. Looks like @dave14305 over at FlexQoS added some code to service-event-end.

See: https://github.com/dave14305/FlexQoS/commit/cae6694cc11dabcf570c3883ff4013e22ba8ab96

Not sure if it has any relevance with this issue, as perhaps a fix in a later update. Looks like there are a few users who are experiencing it and would be happy to test. Let me know what you think.

Adamm00, 2020-07-22:
I still can't reproduce this issue on my end, so there must be other factors at play.

ttgapers, 2020-11-25:
Closing as this has come up multiple times with no ability to reproduce. Current checks will restart cake if for some reason it dies.

closed 2020-11-25 by ttgapers