ttgapers/cakeqos-merlin #60: AX58U/AX3000 don't use eth0 for WAN interface

dave14305, 2020-12-02:
They use eth4.

Adamm00, 2020-12-02:
As I have no way to test, would the interface be the result of wan0_ifname?

dave14305, 2020-12-02:
I recently incorporated this function from Merlin to determine the wan ifname even if dual-wan is in place. Seems to work well.

https://github.com/dave14305/FlexQoS/commit/f6f9668f3a3372b31579e69a3d87d39141e46122

Adamm00, 2020-12-02:
Good function, but not sure if it would be worth adding for the reason if someone lets say has a 100/40 primary connection and 10/1 secondary, the speed limit values being applied to their secondary would be completely inaccurate.

Thoughts?

dave14305, 2020-12-02:
True, but for purposes of cakeqos, it gives you a device-agnostic way to get the current interface. I actually only use it for the firewall portion of my script (you could use it to provide Dual-WAN support for Skynet).

Since the AX58/3000U is the only model like this so far, you could just set a wan variable to eth4 if those models are detected, otherwise eth0.

jata1, 2020-12-02:
Thanks for reporting and looking into this issue. I'm happy to test/help as I have a AX3000 and have manually changed the script to use eth4. Cake now working perfectly...

Adamm00, 2020-12-03:
> True, but for purposes of cakeqos, it gives you a device-agnostic way to get the current interface. I actually only use it for the firewall portion of my script (you could use it to provide Dual-WAN [...]

I'll have to think this one through a little more, I can see an issue arising in dual-wan mode (for both cake and Skynet) that old rules won't be removed because the delete commands will reference the old interface.

dave14305, 2020-12-03:
> I'll have to think this one through a little more, I can see an issue arising in dual-wan mode (for both cake and Skynet) that old rules won't be removed because the delete commands will reference [...]

My personal opinion is to not assume anything when stopping cake and query the interface names from existing cake qdiscs.
`/opt/sbin/tc qdisc | grep "qdisc cake " | awk ' { print $5 } ' `

Adamm00, 2020-12-03:
Let me know if the latest commit works as a temporary solution. I'll address this further when I have some free time to fire up my AX88U for testing as I am currently mid way through a data migration / raid rebuild and can't mess around with my network.

jata1, 2020-12-03:
Thanks. Tested and working correctly now on AX58U/AX3000.

@Adamm00 big thanks for helping with this and hope the migration / rebuild goes well

T-Birth, 2020-12-05:
Tested and working without manually changing eth0 to eth4. Thanks.

ttgapers, 2020-12-08:
@Adamm00 & @dave14305 - safe to close this one now?

Adamm00, 2020-12-15:
Will get to this one eventually, been working 12hr days at a new job so my spare time until Christmas is non existent.

ttgapers, 2020-12-28:
> 
> 
> Will get to this one eventually, been working 12hr days at a new job so my spare time until Christmas is non existent.

No worries man. All good. Trust me....I know how it is!

jata1, 2021-01-28:
Hi all - just checking to see if this fix might not work with merlin 386.1. I have done a new/clean install of 386.1 and cake is not working at all with this build - fails to start but nothing  in the logs

ttgapers, 2021-01-28:
> 
> 
> Hi all - just checking to see if this fix might not work with merlin 386.1. I have done a new/clean install of 386.1 and cake is not working at all with this build - fails to start but nothing in the logs

It still is on the list and as @Adamm00 mentioned as soon as time permits it will be looked at.

jata1, 2021-01-29:
Great stuff and thanks - apologies for cross posting. 

Happy to help test when needed/helpful but I have just gone back to 384.19 as it is very stable and cake working great!

underd0se, 2021-01-29:
Moving my message from SNBforums to here: 

AX58U user here and I have the same symptoms using CakeQoS (no WAN connection, uninstall and reboot to solve the problem). 

wan0_ifname is `et4`, tc qdisc ls is below but before you dive in I have to warn you I don't have CakeQoS installed for the obvious reasons, instead I have FlexQoS so I'm not sure if the output below is helpful or not. 

`qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc htb 1: dev eth4 root refcnt 2 r2q 10 default 0 direct_packets_stat 5023009 direct_qlen 1000
qdisc fq_codel 8015: dev eth4 parent 1:2 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8017: dev eth4 parent 1:10 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8019: dev eth4 parent 1:11 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 801b: dev eth4 parent 1:12 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 801d: dev eth4 parent 1:13 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 801f: dev eth4 parent 1:14 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8021: dev eth4 parent 1:15 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8023: dev eth4 parent 1:16 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8025: dev eth4 parent 1:17 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc pfifo_fast 0: dev eth5 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth6 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc htb 1: dev br0 root refcnt 2 r2q 10 default 0 direct_packets_stat 8862009 direct_qlen 2
qdisc fq_codel 8014: dev br0 parent 1:2 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8016: dev br0 parent 1:10 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8018: dev br0 parent 1:11 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 801a: dev br0 parent 1:12 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 801c: dev br0 parent 1:13 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 801e: dev br0 parent 1:14 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8020: dev br0 parent 1:15 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8022: dev br0 parent 1:16 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
qdisc fq_codel 8024: dev br0 parent 1:17 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn`

tekneek, 2021-01-31:
As 386.1 was officially released yesterday, I upgraded and encountered the same issues previously reported. 

The following is from after I removed cake-qos, as I had to remove it and reboot the router to pass Internet traffic.

```
# tc qdisc ls
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth4 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth5 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth6 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev wl0.1 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev wl0.2 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev wl1.1 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev wl1.2 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1

# nvram get wan0_ifname
eth4
```

 If there is any additional debugging needed, I am willing to install it and collect any output needed (outside of regular business hours (M-F 9-5) - since it will cause a complete outage).

jata1, 2021-02-03:
I have completed a full/clean factory reset and manual setup of my network - all on 386.1. 

I have not setup any QOS yet (stock, Flex or cake). I can confirm that the output from tc qdisc ls is exactly the same as reported above by tekneek. Also happy to help test when needed....

I have swap, jffs, entware and nextdns installed and everything is working perfectly. Overall I am very happy with merlin 386.1 - it is working better than 384.19.

ttgapers, 2021-03-02:
> I have completed a full/clean factory reset and manual setup of my network - all on 386.1.
>
> I have not setup any QOS yet (stock, Flex or cake). I can confirm that the output from tc qdisc ls is [...]

Hi @jata1, have you given cake a test drive since updating? I don't have a device to test with but curious to know as you reported success earlier with the fixes from @Adamm00.

ttgapers, 2021-03-02:
> As 386.1 was officially released yesterday, I upgraded and encountered the same issues previously reported.
>
> The following is from after I removed cake-qos, as I had to remove it and reboot the [...]

Will track based on @jata1 response

ttgapers, 2021-03-02:
> 
> 
> Moving my message from SNBforums to here:
> 
> AX58U user here and I have the same symptoms using CakeQoS with Merlin 386.1 beta 5 (no WAN connection, uninstall and reboot to solve the problem).
> 
> wan0_ifname is `eth4`, tc qdisc ls is below but before you dive in I have to warn you I don't have CakeQoS installed for the obvious reasons, instead I have FlexQoS so I'm not sure if the output below is helpful or not.
> 
> qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
> qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
> qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
> qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
> qdisc htb 1: dev eth4 root refcnt 2 r2q 10 default 0 direct_packets_stat 5023009 direct_qlen 1000
> qdisc fq_codel 8015: dev eth4 parent 1:2 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8017: dev eth4 parent 1:10 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8019: dev eth4 parent 1:11 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 801b: dev eth4 parent 1:12 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 801d: dev eth4 parent 1:13 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 801f: dev eth4 parent 1:14 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8021: dev eth4 parent 1:15 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8023: dev eth4 parent 1:16 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8025: dev eth4 parent 1:17 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc pfifo_fast 0: dev eth5 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
> qdisc pfifo_fast 0: dev eth6 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
> qdisc htb 1: dev br0 root refcnt 2 r2q 10 default 0 direct_packets_stat 8862009 direct_qlen 2
> qdisc fq_codel 8014: dev br0 parent 1:2 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8016: dev br0 parent 1:10 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8018: dev br0 parent 1:11 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 801a: dev br0 parent 1:12 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 801c: dev br0 parent 1:13 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 801e: dev br0 parent 1:14 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8020: dev br0 parent 1:15 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8022: dev br0 parent 1:16 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> qdisc fq_codel 8024: dev br0 parent 1:17 limit 1024p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn
> 
> I can help you test today and over the weekend btw 🤙

Will track based on @jata1 response

jata1, 2021-03-02:
There is s branched cake build created by @underd0se for the AX58U running on 386.1. Might be worth catching up on the merlin add-on thread where we discussed it. 

https://www.snbforums.com/threads/cake-not-working-with-ax58u-on-386-1.69720/

In summary, the fix was to replace the binaries with the newer ones from Odkrys' repo. I have cake running perfectly on my RT-AX58U so I think all that needs to happen are for the main cake branch to pull/merge with these changes to replace the binaries.

I'm sure @underd0se can chip into this discussion

jata1, 2021-03-02:
Forgot to say... that the changes made to the script a while back by @Adamm00 are working on my AX3000 and AX58U (as I have both models)... These script changes fixed the eth4 v eth0 issue but do not address the recent issue with 386.1

ttgapers, 2021-03-02:
> There is s branched cake build created by @underd0se for the AX58U running on 386.1. Might be worth catching up on the merlin add-on thread where we discussed it. [...]

Trust me -- this may not be necessary soon enough. Will reply there just so peeps know we are tracking this.

jata1, 2021-03-02:
@ttgapers - Thanks. Have just caught up with the Merlin Alpha thread for 386.2 with cake built-in. That's great news. Do you know if merlin is aware of the issue with eth4 v eth0 on AX58U?

ttgapers, 2021-03-04:
> 
> 
> @ttgapers - Thanks. Have just caught up with the Merlin Alpha thread for 386.2 with cake built-in. That's great news. Do you know if merlin is aware of the issue with eth4 v eth0 on AX58U?

Yup and will not be an issue as per the Wizard. Will leave this open until Adamm or I get around to pushing this to the current branch.

ttgapers, 2021-03-04:
Hi,

As mentioned in the other thread, I've committed updates to the develop branch that includes the changes from @Adamm00 as well as the new cake binaries.

`mkdir -p /jffs/addons/cake-qos && /usr/sbin/curl -s "https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/develop/cake-qos.sh" -o "/jffs/addons/cake-qos/cake-qos" && chmod 755 /jffs/addons/cake-qos/cake-qos && sh /jffs/addons/cake-qos/cake-qos install`

tekneek, 2021-03-04:
Been using this for approximately the past couple of hours without any problems.

ttgapers, 2021-03-05:
Fixed and committed to version v1.0.8

closed 2021-03-05 by ttgapers