ttgapers/cakeqos-merlin #61: AX58U/AX3000 - runner: not found error when stopping cake-qos

jata1, 2020-12-02:
when stopping cake using cake-qos stop the following error is returned

/opt/bin/cake-qos: line 745: runner: not found

I see that both runner and flow cache are both enabled (in GUI) after stopping cake so it is probably working correctly. Not sure if the NVRAM setting is set and applied though?

Adamm00, 2020-12-03:
> when stopping cake using cake-qos stop the following error is returned
> 
> /opt/bin/cake-qos: line 745: runner: not found
> 
> I see that both runner and flow cache are both enabled (in GUI) after stopping cake so it is probably working correctly. Not sure if the NVRAM setting is set and applied though?

Can you run the following command and post the output;

`sh -x /jffs/addons/cake-qos/cake-qos stop`

jata1, 2020-12-03:
+ clear
+ sed -n 6,21p /jffs/addons/cake-qos/cake-qos

#########################################################
##               _                                     ##
##              | |                                    ##
##    ___  __ _ | | __ ___          __ _   ___   ___   ##
##   / __|/ _` || |/ // _ \ ______ / _` | / _ \ / __|  ##
##  | (__ |(_| ||   <|  __/|______| (_| || (_) |\__ \  ##
##   \___|\__,_||_|\_\\___|        \__, | \___/ |___/  ##
##                                    | |              ##
##                                    |_|              ##
##                                                     ##
##      https://github.com/ttgapers/cakeqos-merlin     ##
##                        v1.0.5                       ##
##                                                     ##
#########################################################

+ readonly SCRIPT_NAME=cake-qos
+ readonly SCRIPT_NAME_FANCY=CakeQOS-Merlin
+ readonly SCRIPT_BRANCH=master
+ readonly SCRIPT_DIR=/jffs/addons/cake-qos
+ readonly SCRIPT_CFG=/jffs/addons/cake-qos/cake-qos.cfg
+ readonly SCRIPT_REMOTEDIR=https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/master
+ readonly CRIT=\e[41m
+ readonly ERR=\e[31m
+ readonly WARN=\e[33m
+ readonly PASS=\e[32m
+ nvram get odmpid
+ [ -z RT-AX3000 ]
+ nvram get odmpid
+ RMODEL=RT-AX3000
+ [ -f /jffs/addons/cake-qos/cake-qos.cfg ]
+ . /jffs/addons/cake-qos/cake-qos.cfg
+ dlspeed=50
+ upspeed=20
+ queueprio=besteffort
+ optionsdl=bridged-ptm
+ optionsup=bridged-ptm
+ nvram get wan0_ifname
+ iface=eth4
+ [ -z stop ]
+ [ -n  ]
+ Display_Line
+ printf \n#########################################################\n\n

#########################################################

+ Cake_Stop
+ Cake_CheckStatus
+ tc qdisc
+ grep -E ^qdisc cake .* dev eth4 root
+ STATUS_UPLOAD=qdisc cake 800b: dev eth4 root refcnt 2 bandwidth 20Mbit besteffort triple-isolate nat nowash no-ack-filter split-gso rtt                                                                                                    100.0ms ptm overhead 22 no-sce
+ tc+ grep -E ^qdisc cake .* dev ifb9eth4 root
 qdisc
+ STATUS_DOWNLOAD=qdisc cake 800c: dev ifb9eth4 root refcnt 2 bandwidth 50Mbit besteffort triple-isolate nat wash ingress no-ack-filter sp                                                                                                   lit-gso rtt 100.0ms ptm overhead 22 no-sce
+ [ -n qdisc cake 800b: dev eth4 root refcnt 2 bandwidth 20Mbit besteffort triple-isolate nat nowash no-ack-filter split-gso rtt 100.0ms p                                                                                                   tm overhead 22 no-sce  ]
+ [ -n qdisc cake 800c: dev ifb9eth4 root refcnt 2 bandwidth 50Mbit besteffort triple-isolate nat wash ingress no-ack-filter split-gso rtt                                                                                                    100.0ms ptm overhead 22 no-sce  ]
+ return 0
+ Print_Output true Stopping \e[32m
+ [ true = true ]
+ logger -t CakeQOS-Merlin Stopping
+ printf \e[1m\e[32m%s: Stopping\e[0m\n CakeQOS-Merlin
CakeQOS-Merlin: Stopping
+ cru d CakeQOS-Merlin
+ /opt/sbin/tc qdisc del dev eth4 ingress
+ /opt/sbin/tc qdisc del dev ifb9eth4 root
+ /opt/sbin/tc qdisc del dev eth4 root
+ ip link del ifb9eth4
+ rmmod sch_cake
+ runner enable
/jffs/addons/cake-qos/cake-qos: line 747: runner: not found
+ fc enable
Broadcom Packet Flow Cache learning via BLOG enabled.
+ nvram set runner_disable=0
+ nvram set fc_disable=0
+ nvram commit
+ tty
+ sed -i \~# CakeQOS-Merlin~d /jffs/scripts/nat-start /jffs/scripts/services-stop
+ Display_Line
+ printf \n#########################################################\n\n

#########################################################

+ [ -n  ]

jata1, 2020-12-03:
`+ clear
+ sed -n 6,21p /jffs/addons/cake-qos/cake-qos

#########################################################
##               _                                     ##
##              | |                                    ##
##    ___  __ _ | | __ ___          __ _   ___   ___   ##
##   / __|/ _` || |/ // _ \ ______ / _` | / _ \ / __|  ##
##  | (__ |(_| ||   <|  __/|______| (_| || (_) |\__ \  ##
##   \___|\__,_||_|\_\\___|        \__, | \___/ |___/  ##
##                                    | |              ##
##                                    |_|              ##
##                                                     ##
##      https://github.com/ttgapers/cakeqos-merlin     ##
##                        v1.0.5                       ##
##                                                     ##
#########################################################

+ readonly SCRIPT_NAME=cake-qos
+ readonly SCRIPT_NAME_FANCY=CakeQOS-Merlin
+ readonly SCRIPT_BRANCH=master
+ readonly SCRIPT_DIR=/jffs/addons/cake-qos
+ readonly SCRIPT_CFG=/jffs/addons/cake-qos/cake-qos.cfg
+ readonly SCRIPT_REMOTEDIR=https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/master
+ readonly CRIT=\e[41m
+ readonly ERR=\e[31m
+ readonly WARN=\e[33m
+ readonly PASS=\e[32m
+ nvram get odmpid
+ [ -z RT-AX3000 ]
+ nvram get odmpid
+ RMODEL=RT-AX3000
+ [ -f /jffs/addons/cake-qos/cake-qos.cfg ]
+ . /jffs/addons/cake-qos/cake-qos.cfg
+ dlspeed=50
+ upspeed=20
+ queueprio=besteffort
+ optionsdl=bridged-ptm
+ optionsup=bridged-ptm
+ nvram get wan0_ifname
+ iface=eth4
+ [ -z stop ]
+ [ -n  ]
+ Display_Line
+ printf \n#########################################################\n\n

#########################################################

+ Cake_Stop
+ Cake_CheckStatus
+ tc qdisc
+ grep -E ^qdisc cake .* dev eth4 root
+ STATUS_UPLOAD=qdisc cake 800b: dev eth4 root refcnt 2 bandwidth 20Mbit besteffort triple-isolate nat nowash no-ack-filter split-gso rtt                                                                                                    100.0ms ptm overhead 22 no-sce
+ tc+ grep -E ^qdisc cake .* dev ifb9eth4 root
 qdisc
+ STATUS_DOWNLOAD=qdisc cake 800c: dev ifb9eth4 root refcnt 2 bandwidth 50Mbit besteffort triple-isolate nat wash ingress no-ack-filter sp                                                                                                   lit-gso rtt 100.0ms ptm overhead 22 no-sce
+ [ -n qdisc cake 800b: dev eth4 root refcnt 2 bandwidth 20Mbit besteffort triple-isolate nat nowash no-ack-filter split-gso rtt 100.0ms p                                                                                                   tm overhead 22 no-sce  ]
+ [ -n qdisc cake 800c: dev ifb9eth4 root refcnt 2 bandwidth 50Mbit besteffort triple-isolate nat wash ingress no-ack-filter split-gso rtt                                                                                                    100.0ms ptm overhead 22 no-sce  ]
+ return 0
+ Print_Output true Stopping \e[32m
+ [ true = true ]
+ logger -t CakeQOS-Merlin Stopping
+ printf \e[1m\e[32m%s: Stopping\e[0m\n CakeQOS-Merlin
CakeQOS-Merlin: Stopping
+ cru d CakeQOS-Merlin
+ /opt/sbin/tc qdisc del dev eth4 ingress
+ /opt/sbin/tc qdisc del dev ifb9eth4 root
+ /opt/sbin/tc qdisc del dev eth4 root
+ ip link del ifb9eth4
+ rmmod sch_cake
+ runner enable
/jffs/addons/cake-qos/cake-qos: line 747: runner: not found
+ fc enable
Broadcom Packet Flow Cache learning via BLOG enabled.
+ nvram set runner_disable=0
+ nvram set fc_disable=0
+ nvram commit
+ tty
+ sed -i \~# CakeQOS-Merlin~d /jffs/scripts/nat-start /jffs/scripts/services-stop
+ Display_Line
+ printf \n#########################################################\n\n

#########################################################

+ [ -n  ]
`

jata1, 2020-12-03:
Hope this helps - having trouble inserting as code. sorry.

`
+ readonly SCRIPT_NAME=cake-qos
+ readonly SCRIPT_NAME_FANCY=CakeQOS-Merlin
+ readonly SCRIPT_BRANCH=master
+ readonly SCRIPT_DIR=/jffs/addons/cake-qos
+ readonly SCRIPT_CFG=/jffs/addons/cake-qos/cake-qos.cfg
+ readonly SCRIPT_REMOTEDIR=https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/master
+ readonly CRIT=\e[41m
+ readonly ERR=\e[31m
+ readonly WARN=\e[33m
+ readonly PASS=\e[32m
+ nvram get odmpid
+ [ -z RT-AX3000 ]
+ nvram get odmpid
+ RMODEL=RT-AX3000
+ [ -f /jffs/addons/cake-qos/cake-qos.cfg ]
+ . /jffs/addons/cake-qos/cake-qos.cfg
+ dlspeed=50
+ upspeed=20
+ queueprio=besteffort
+ optionsdl=bridged-ptm
+ optionsup=bridged-ptm
+ nvram get wan0_ifname
+ iface=eth4
+ [ -z stop ]
+ [ -n  ]
+ Display_Line
+ printf \n#########################################################\n\n

#########################################################

+ Cake_Stop
+ Cake_CheckStatus
+ tc qdisc
+ grep -E ^qdisc cake .* dev eth4 root
+ STATUS_UPLOAD=qdisc cake 800b: dev eth4 root refcnt 2 bandwidth 20Mbit besteffort triple-isolate nat nowash no-ack-filter split-gso rtt                                                                                                    100.0ms ptm overhead 22 no-sce
+ tc+ grep -E ^qdisc cake .* dev ifb9eth4 root
 qdisc
+ STATUS_DOWNLOAD=qdisc cake 800c: dev ifb9eth4 root refcnt 2 bandwidth 50Mbit besteffort triple-isolate nat wash ingress no-ack-filter sp                                                                                                   lit-gso rtt 100.0ms ptm overhead 22 no-sce
+ [ -n qdisc cake 800b: dev eth4 root refcnt 2 bandwidth 20Mbit besteffort triple-isolate nat nowash no-ack-filter split-gso rtt 100.0ms p                                                                                                   tm overhead 22 no-sce  ]
+ [ -n qdisc cake 800c: dev ifb9eth4 root refcnt 2 bandwidth 50Mbit besteffort triple-isolate nat wash ingress no-ack-filter split-gso rtt                                                                                                    100.0ms ptm overhead 22 no-sce  ]
+ return 0
+ Print_Output true Stopping \e[32m
+ [ true = true ]
+ logger -t CakeQOS-Merlin Stopping
+ printf \e[1m\e[32m%s: Stopping\e[0m\n CakeQOS-Merlin
CakeQOS-Merlin: Stopping
+ cru d CakeQOS-Merlin
+ /opt/sbin/tc qdisc del dev eth4 ingress
+ /opt/sbin/tc qdisc del dev ifb9eth4 root
+ /opt/sbin/tc qdisc del dev eth4 root
+ ip link del ifb9eth4
+ rmmod sch_cake
+ runner enable
/jffs/addons/cake-qos/cake-qos: line 747: runner: not found
+ fc enable
Broadcom Packet Flow Cache learning via BLOG enabled.
+ nvram set runner_disable=0
+ nvram set fc_disable=0
+ nvram commit
+ tty
+ sed -i \~# CakeQOS-Merlin~d /jffs/scripts/nat-start /jffs/scripts/services-stop
+ Display_Line
+ printf \n#########################################################\n\n

#########################################################

+ [ -n  ]

`

Adamm00, 2020-12-04:
> runner enable
> /jffs/addons/cake-qos/cake-qos: line 747: runner: not found

So apparently the runner command doesn't exist on this platform. What is the output of;

`ls /bin`

T-Birth, 2020-12-05:
currently having the same issue on my AX-58U and here are the results for ->

sh -x /jffs/addons/cake-qos/cake-qos stop

```
+ sed -n 6,21p /jffs/addons/cake-qos/cake-qos

#########################################################
##               _                                     ##
##              | |                                    ##
##    ___  __ _ | | __ ___          __ _   ___   ___   ##
##   / __|/ _` || |/ // _ \ ______ / _` | / _ \ / __|  ##
##  | (__ |(_| ||   <|  __/|______| (_| || (_) |\__ \  ##
##   \___|\__,_||_|\_\\___|        \__, | \___/ |___/  ##
##                                    | |              ##
##                                    |_|              ##
##                                                     ##
##      https://github.com/ttgapers/cakeqos-merlin     ##
##                        v1.0.5                       ##
##                                                     ##
#########################################################

+ readonly SCRIPT_NAME=cake-qos
+ readonly SCRIPT_NAME_FANCY=CakeQOS-Merlin
+ readonly SCRIPT_BRANCH=master
+ readonly SCRIPT_DIR=/jffs/addons/cake-qos
+ readonly SCRIPT_CFG=/jffs/addons/cake-qos/cake-qos.cfg
+ readonly SCRIPT_REMOTEDIR=https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/master
+ readonly CRIT=\e[41m
+ readonly ERR=\e[31m
+ readonly WARN=\e[33m
+ readonly PASS=\e[32m
+ nvram get odmpid
+ [ -z  ]
+ nvram get productid
+ RMODEL=RT-AX58U
+ [ -f /jffs/addons/cake-qos/cake-qos.cfg ]
+ . /jffs/addons/cake-qos/cake-qos.cfg
+ dlspeed=73
+ upspeed=34
+ queueprio=besteffort
+ optionsdl=ethernet nowash regional
+ optionsup=ethernet ack-filter regional
+ nvram get wan0_ifname
+ iface=eth4
+ [ -z stop ]
+ [ -n  ]
+ Display_Line
+ printf \n#########################################################\n\n

#########################################################

+ Cake_Stop
+ Cake_CheckStatus
+ tc qdisc
+ grep -E ^qdisc cake .* dev eth4 root
+ STATUS_UPLOAD=qdisc cake 8001: dev eth4 root refcnt 2 bandwidth 34Mbit besteffort triple-isolate nat nowash ack-filter split-gso rtt 30.0ms noatm overhead 38 mpu 84 no-sce 
+ tc qdisc
+ grep -E ^qdisc cake .* dev ifb9eth4 root
+ STATUS_DOWNLOAD=qdisc cake 8003: dev ifb9eth4 root refcnt 2 bandwidth 73Mbit besteffort triple-isolate nat nowash ingress no-ack-filter split-gso rtt 30.0ms noatm overhead 38 mpu 84 no-sce 
+ [ -n qdisc cake 8001: dev eth4 root refcnt 2 bandwidth 34Mbit besteffort triple-isolate nat nowash ack-filter split-gso rtt 30.0ms noatm overhead 38 mpu 84 no-sce  ]
+ [ -n qdisc cake 8003: dev ifb9eth4 root refcnt 2 bandwidth 73Mbit besteffort triple-isolate nat nowash ingress no-ack-filter split-gso rtt 30.0ms noatm overhead 38 mpu 84 no-sce  ]
+ return 0
+ Print_Output true Stopping \e[32m
+ [ true = true ]
+ logger -t CakeQOS-Merlin Stopping
+ printf \e[1m\e[32m%s: Stopping\e[0m\n CakeQOS-Merlin
CakeQOS-Merlin: Stopping
+ cru d CakeQOS-Merlin
+ /opt/sbin/tc qdisc del dev eth4 ingress
[... 21 lines cut ...]

Output of ls /bin ->
```

DoctorWho1337@RT-AX58U-41D0:/tmp/home/root# ls /bin
archer             chmod              ethswctl           grep               more               ps                 stress             ubiupdatevol
archerctl          chown              false              gunzip             mount              psictl             swmdk              umount
ash                comgt              fc                 gzip               mtd_debug          pspctl             sync               uname
bash               cp                 fcctl              hostname           mtdinfo            pwd                tar                urlfilterd
bcm_boot_launcher  date               fgrep              ionice             mtdpart            pwr                tmctl              usleep
bcm_bootstate      dd                 flash_erase        kill               mv                 pwrctl             touch              vi
bcm_flasher        df                 flash_eraseall     ln                 nanddump           rawSocketTest      true               vlanctl
bcmmcastctl        dmesg              flash_lock         login              nandtest           recv_image         ubiattach          vpmstats
bcmtm              dnsdomainname      flash_otp_dump     ls                 nandwrite          rfddump            ubiblock           watch
bcmtmctl           dnsspoof           flash_otp_info     lsmtd              netstat            rfdformat          ubicrc32           wdtctl
bmc                doc_loadbios       flash_otp_lock     makemime           nftl_format        rm                 ubidetach          wlaffinity
bpm                docfdisk           flash_otp_write    mcp                nftldump           rmdir              ubiformat          wps_monitor
bpmctl             dry                flash_unlock       mcpctl             nice               rstats             ubimkvol           zcat
brctl              eapd               flashcp            mcpd               nvram              scratchpadctl      ubinfo
busybox            echo               fsync              mdkshell           nvramUpdate        sed                ubinize
cat                egrep              ftl_check          memaccess          pidof              serve_image        ubirename
ceventc            epi_ttcp           ftl_format         mkdir              ping               sh                 ubirmvol
ceventd            ethctl             getopt             mknod              ping6              sleep              ubirsvol
DoctorWho1337@RT-AX58U-41D0:/tmp/home/root# 
```

Adamm00, 2020-12-06:
Looks like the runner binary (or runner in general) isn't available on the AX58U, you would have to contact Merlin and find out why.

T-Birth, 2020-12-06:
`RMerlin: The bcm675x does not use runner, that's for the bcm490x.`

jata1, 2020-12-06:
@T-Birth - did merlin say how runner HW acceleration works on AX58U as it does show it in the tools section of the FW?

[image]

dave14305, 2020-12-07:
So it looks like the syntax for the AX58/3000U in place of `runner disable` might be:
`fc config --hw-accel 0`

https://github.com/RMerl/asuswrt-merlin.ng/blob/c514f9f0b3f279dc1ea91ede4c92ab3e02beb8fa/release/src/router/rc/sysdeps/init-broadcom.c#L9468

Adamm00, 2020-12-07:
Thanks @dave14305 , if we can get someone to confirm the command works I'll go ahead and throw in an if statement.

jata1, 2020-12-07:
Just looked into this and here is what I have found:

1. script is already working to disable/enable Flow Cache when cake starts and stops
2. fc config --hw-accel 0 (and 1) does the same as the script

So I think that the issue is mainly cosmetic - i.e. that the GUI in merlin (and maybe NVRAM settings) includes runner and flow cache settings but the AX-58U does not have runner HW acceleration at all?

If this is true, then i think the fix is to exclude the runner enable/disable code in the script for this model

Adamm00, 2020-12-07:
Going to close this for now, feel free to reopen if there's any issues with v1.0.6

closed 2020-12-07 by Adamm00