ttgapers/cakeqos-merlin #62: Cake/TC need to be recompiled for AX86U

Adamm00, 2020-12-03:
Ideally with some instructions on how it was compiled in the first place so this task can be automated for future updates / models.

ttgapers, 2020-12-08:
Hi @Adamm00 

Looks like @odkrys is the member providing the builds and might be able to assist on this one. I would really like to see collab on this one. Here is the link to his repo: https://github.com/odkrys/entware-makefile-for-merlin

dave14305, 2020-12-09:
I have some ideas on setting this up since I’ve been successfully building my own openwrt images, and Entware uses the openwrt buildroot. I am thinking this has to have 1) openwrt buildroot, 2) Entware repo, 3) asuswrt-merlin.ng repo, and 4) sched_cake_oot repo all available to build. Will test it out and let you know if I’m full of it (hint: probably).

Adamm00, 2020-12-09:
> I have some ideas on setting this up since I’ve been successfully building my own openwrt images, and Entware uses the openwrt buildroot. I am thinking this has to have 1) openwrt buildroot, 2) [...]

Yeah I tried briefly last week and gave up, I'm sure I was missing a step somewhere. In any case if you figure it out I can automate it for future updates/models.

dave14305, 2020-12-09:
> Yeah I tried briefly last week and gave up, I'm sure I was missing a step somewhere. In any case if you figure it out I can automate it for future updates/models.

I’ve given up for now since I couldn’t build the Entware toolchain on WSL2 Ubuntu. Something wasn’t set right and couldn’t figure it out. Told you I was full of it.

ttgapers, 2020-12-28:
Thanks @dave14305 -- would really like to get @odkrys involved for his feedback/help with this.

shemminger, 2021-01-16:
Does this version of cake support ack-filtering?
I have 1G down (in theory) but only 10M up this large asymmetric link can really benefit from ack compression.

dave14305, 2021-02-02:
> 
> 
> Thanks @dave14305 -- would really like to get @odkrys involved for his feedback/help with this.

Looks like he compiled new builds today. Will anyone test them?

tekneek, 2021-02-03:
What is required to help out w/ testing? I looked at the repo, but not obvious (to me) what I would need to do from my end. Is the process documented?

underd0se, 2021-02-03:
> > Thanks @dave14305 -- would really like to get @odkrys involved for his feedback/help with this.
> 
> Looks like he compiled new builds today. Will anyone test them?

I've forked CakeQoS and made necessary changes, run the script and it works like a charm on my RT-AX58U.

dave14305, 2021-02-03:
> qdisc cake 801b: dev eth4 root refcnt 2 [Unknown qdisc, optlen=140]

It doesn’t look like tc-adv is working for you. It shouldn’t be Unknown qdisc.

ttgapers, 2021-02-03:
> 
> 
> What is required to help out w/ testing? I looked at the repo, but not obvious (to me) what I would need to do from my end. Is the process documented?

That's the problem I see, no documentation anywhere, so left looking at the Makefile.  You need a kernel to build against, and @odkrys uses Merlin's new builds. Second are these two includes:

`include $(TOPDIR)/rules.mk
include $(INCLUDE_DIR)/kernel.mk`

sched-cake-oot: [https://github.com/odkrys/entware-makefile-for-merlin/blob/main/sched-cake-oot/Makefile](url)
tc-adv: [https://github.com/odkrys/entware-makefile-for-merlin/blob/main/tc-adv/Makefile](url)

Seems @odkrys is building cake into the Merlin kernel natively...which is also pretty cool 💯

underd0se, 2021-02-03:
> > qdisc cake 801b: dev eth4 root refcnt 2 [Unknown qdisc, optlen=140]
> 
> It doesn’t look like tc-adv is working for you. It shouldn’t be Unknown qdisc.

DSLReports' Speedtest scores went from B C B to A+ A A+, so there's definitely an improvement there 🤔 I am nowhere near a developer but is it possible that "_Unknown_ qdisc" is a cosmetic error caused by an undefined string or something and the script just works? 

I've made several tests in the last 2 weeks on this unit with all the QoS options (Traditional, Adaptive, FlexQoS) and never got any A's on DSLReports, that's why I'm asking.

dave14305, 2021-02-03:
> > > qdisc cake 801b: dev eth4 root refcnt 2 [Unknown qdisc, optlen=140]
> > 
> > 
> > It doesn’t look like tc-adv is working for you. It shouldn’t be Unknown qdisc.
> 
> DSLReports' Speedtest scores went from B C B to A+ A A+, so there's definitely an improvement there 🤔 I am nowhere near a developer but is it possible that "_Unknown_ qdisc" is a cosmetic error caused by an undefined string or something and the script just works?
> 
> I've made several tests in the last 2 weeks on this unit with all the QoS options (Traditional, Adaptive, FlexQoS) turned on and off, and never got any A's on DSLReports, that's why I'm asking.

Try running `/opt/sbin/tc -s qdisc`

underd0se, 2021-02-03:
@dave14305 

/opt/bin/tc -s qdisc

```
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 4881881 bytes 16347 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc cake 8021: dev eth4 root refcnt 2 bandwidth 5Mbit diffserv4 triple-isolate nat nowash no-ack-filter split-gso rtt 100.0ms noatm overhead 18 mpu 64 no-sce
 Sent 2095617463 bytes 6902050 pkt (dropped 42129, overlimits 6022502 requeues 0)
 backlog 0b 0p requeues 0
 memory used: 4195776b of 4Mb
 capacity estimate: 5Mbit
 min/max network layer size:           28 /    1500
 min/max overhead-adjusted size:       64 /    1518
 average network hdr offset:           14

                   Bulk  Best Effort        Video        Voice
  thresh      312496bit        5Mbit     2500Kbit     1250Kbit
  target         58.1ms        5.0ms        7.3ms       14.5ms
  interval      153.1ms      100.0ms      102.3ms      109.5ms
  pk_delay          0us        7.0ms        2.8ms        2.1ms
  av_delay          0us        1.7ms         76us        642us
  sp_delay          0us         94us         17us         17us
  backlog            0b           0b           0b           0b
  pkts                0      6942812           30         1337
  bytes               0   2137932268        13412       188596
  way_inds            0      1480256            0            0
  way_miss            0      2008637            2           43
  way_cols            0            4            0            0
  marks               0            0            0            0
  drops               0        42129            0            0
  ack_drop            0            0            0            0
  sp_flows            0           10            1            1
  bk_flows            0            1            0            0
  un_flows            0            0            0            0
  max_len             0         1514         1422          590
  quantum           300          300          300          300

qdisc ingress ffff: dev eth4 parent ffff:fff1 ----------------
 Sent 7298468147 bytes 7045450 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc pfifo_fast 0: dev eth5 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 5024623589 bytes 5479014 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc pfifo_fast 0: dev eth6 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 4243992972 bytes 4262384 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc cake 8022: dev ifb9eth4 root refcnt 2 bandwidth 95Mbit diffserv4 triple-isolate nat wash ingress no-ack-filter split-gso rtt 100.0ms noatm overhead 18 mpu 64 no-sce
 Sent 7396053500 bytes 7044743 pkt (dropped 704, overlimits 3470051 requeues 0)
 backlog 0b 0p requeues 0
 memory used: 996Kb of 4750000b
 capacity estimate: 95Mbit
 min/max network layer size:           46 /    1500
 min/max overhead-adjusted size:       64 /    1518
 average network hdr offset:           14

                   Bulk  Best Effort        Video        Voice
  thresh       5937Kbit       95Mbit    47500Kbit    23750Kbit
  target          5.0ms        5.0ms        5.0ms        5.0ms
  interval      100.0ms      100.0ms      100.0ms      100.0ms
  pk_delay          0us        105us        1.2ms         33us
  av_delay          0us         11us        199us          6us
  sp_delay          0us          4us         34us          3us
  backlog            0b           0b           0b           0b
  pkts                0      6803483       236514         5450
  bytes               0   7088573993    308202830       327426
  way_inds            0       593764            7            0
  way_miss            0       324896           46            4
  way_cols            0            0            0            0
  marks               0          235            0            0
  drops               0          696            8            0
  ack_drop            0            0            0            0
  sp_flows            0            1            1            1
  bk_flows            0            1            0            0
  un_flows            0            0            0            0
  max_len             0         1514         1446          182
  quantum           300         1514         1449          724
```

ttgapers, 2021-02-03:
> I’ve given up for now since I couldn’t build the Entware toolchain on WSL2 Ubuntu. Something wasn’t set right and couldn’t figure it out. Told you I was full of it.

@dave14305 

Can you document what you have done to get to the point of failure? I have a couple Ubuntu VMs hanging around that I wouldn't mind trying on....

ttgapers, 2021-03-04:
> 
> 
> Does this version of cake support ack-filtering?
> I have 1G down (in theory) but only 10M up this large asymmetric link can really benefit from ack compression.

Sorry for the late reply. Yes it does support ack-filtering. Once you are ok with loosing a fair bit of your down cake on the router should be ok.

ttgapers, 2021-03-04:
@underd0se @tekneek and others here I've just uploaded some changes to the develop branch using the binaries from @odkrys

Apologies on the massive delays. Day (and night) work sometimes gets in the way. 

`mkdir -p /jffs/addons/cake-qos && /usr/sbin/curl -s "https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/develop/cake-qos.sh" -o "/jffs/addons/cake-qos/cake-qos" && chmod 755 /jffs/addons/cake-qos/cake-qos && sh /jffs/addons/cake-qos/cake-qos install`

Please let me know your feedback, and we get it pushed to master.

This mod is going to phase into an advanced addon for Merlin's firmware with it being rolled into the firmware, which is excellent as we don't have to worry about new builds. I'm very thankful for that as it will hopefully increase adoption of this awesome QoS alternative (with privacy). Testing in the new Alpha has been going good thus far.

underd0se, 2021-03-04:
I've switched to your dev branch, installed cake-qos and: 

1. cake-qos menu doesn't work
None of the CLI menu items seem to work, except uninstall (luckily). 

2. Internet connection is blocked until removal & reboot 

Now I'm back on mine since me and wife are WFH and we need a stable connection. However, I can test it out on the weekend if you make changes.

ttgapers, 2021-03-04:
> I've switched to your dev branch, installed cake-qos and:
>
>     1. cake-qos menu doesn't work
>        None of the CLI menu items seem to work, except uninstall (luckily).
>
>     2. Internet connection [...]

Ok thanks for the quick update. I'll check the diffs in your changes to see if anything was missed.

ttgapers, 2021-03-04:
> I've switched to your dev branch, installed cake-qos and:
>
>     1. cake-qos menu doesn't work
>        None of the CLI menu items seem to work, except uninstall (luckily).
>
>     2. Internet connection [...]

Should be good now. I forgot to update the branch variable in the script. Apologies and thanks for testing again.

ttgapers, 2021-03-05:
Fixed and committed to version v1.0.8

closed 2021-03-05 by ttgapers