ttgapers/cakeqos-merlin #66: AX58U No Internet

coffeevotary, 2020-12-31:
Install AX58U via Entware works
When enabling Cake-QOS, all internet traffic is stopped. LAN traffic works.  Any setting within Cake-QOS, no go. 

No error message found in system logs.
To re-enable internet, must uninstall Cake-QOS and restart router.

ghost-420, 2021-01-06:
I have the same issue on the AX58U running Merlin 386.1_beta3 and CakeQOS-Merlin 1.0.7.
Might edit later if I find any new logs after I test again.

jata1, 2021-01-28:
I have the same issue with cake but only with 386.1 (b5)

ttgapers, 2021-01-29:
> 
> 
> Install AX58U via Entware works
> When enabling Cake-QOS, all internet traffic is stopped. LAN traffic works. Any setting within Cake-QOS, no go.
> 
> No error message found in system logs.
> To re-enable internet, must uninstall Cake-QOS and restart router.

closed 2021-01-29 by ttgapers