ttgapers/cakeqos-merlin #69: cake-qos Cannot find device "root" / Unknown qdisc "ffff:", hence option "ingress" is unparsable / Cannot find device "parent"

raoulteeuwen, 2021-02-28:
**Describe the bug**
Upon CakeQos start, i get 

#########################################################
##               _                                     ##
##              | |                                    ##
##    ___  __ _ | | __ ___          __ _   ___   ___   ##
##   / __|/ _` || |/ // _ \ ______ / _` | / _ \ / __|  ##
##  | (__ |(_| ||   <|  __/|______| (_| || (_) |\__ \  ##
##   \___|\__,_||_|\_\\___|        \__, | \___/ |___/  ##
##                                    | |              ##
##                                    |_|              ##
##                                                     ##
##      https://github.com/ttgapers/cakeqos-merlin     ##
##                        v1.0.7                       ##
##                                                     ##
#########################################################

#########################################################

CakeQOS-Merlin: Starting - ( 50Mbit | 50Mbit | besteffort |  |  )
Broadcom Packet Flow Cache learning via BLOG disabled.
Broadcom Packet Flow Cache flushing the flows
Cannot find device "root"
RTNETLINK answers: File exists
Unknown qdisc "ffff:", hence option "ingress" is unparsable
Cannot find device "parent"

This is on a RT-AC86U running Merlin 386.1_2

**To Reproduce**
Steps to reproduce the behavior:
SSH into machine, execute cake-qos start

**Expected behavior**
cake-qos starts successfully

**Screenshots**
-

**Desktop (please complete the following information):**
 - OS: Win10

**Additional context**
Relative n00b here, so feel free to give up/don't provide (more) help. Could it be my usb stick (ext4 formatted) does not have sufficient capacity (is it needed during runtime, or only during install)? Fyi, upon install, i did leave options for download and upload blank (did not see any hints about what to enter at those points during install, so hoped leaving them blank would be ok).

ttgapers, 2021-03-02:
Hi

You need to add download/upload speeds. It's the 3rd tip [here](https://github.com/ttgapers/cakeqos-merlin#tips).

raoulteeuwen, 2021-03-02:
I think i did: when going through the prompts, i specified 50 Mbit as up & down. Why do you think THAT is/was the problem? Why would that lead to messages like "Cannot find device "root"", "Unknown qdisc "ffff:", hence option "ingress" is unparsable" and "Cannot find device "parent""?

dave14305, 2021-03-02:
What is the output of these commands on your router?
`nvram get wan0_ifname
nvram get wan0_pppoe_ifname
tc qdisc ls`

Do you have a ppp connection or some other Dual-WAN setup?

raoulteeuwen, 2021-03-02:
nvram get wan0_ifname returns nothing
nvram get wan0_pppoe_ifname returns nothing
tc qdisc ls returns
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth4 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth5 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth6 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc cake 803b: dev ifb9 root refcnt 2 bandwidth 50Mbit besteffort triple-isolate nat wash ingress no-ack-filter split-gso rtt 100.0ms raw overhead 0 no-sce

ttgapers, 2021-03-02:
> 
> 
> I think i did: when going through the prompts, i specified 50 Mbit as up & down. Why do you think THAT is/was the problem? Why would that lead to messages like "Cannot find device "root"", "Unknown qdisc "ffff:", hence option "ingress" is unparsable" and "Cannot find device "parent""?

Sorry I missed your up/down settings when replying last night. Do @dave14305 ques, what type of internet connection do you currently have?

raoulteeuwen, 2021-03-02:
> Sorry I missed your up/down settings when replying last night. Do @dave14305 ques, what type of internet connection do you currently have?

No prob, tx for all you do. I have fiber to the home.

dave14305, 2021-03-03:
> nvram get wan0_ifname returns nothing
> nvram get wan0_pppoe_ifname returns nothing

Very odd, what about:
`nvram show | grep ifnam

raoulteeuwen, 2021-03-03:
That gives:

acs_ifnames=
size: 69397 bytes (61675 left)
br0_ifname=br0
br0_ifnames=eth0 eth1 eth2 eth3 eth4 eth5 eth6
bsd_ifnames=eth5 eth6
dpsta_all_ifnames=eth5 eth6
dpsta_ifnames=
eth_ifnames=eth0
lan_ifname=br0
lan_ifnames=eth0 eth1 eth2 eth3 eth4 eth5 eth6
printer_ifname=usb
sta_ifnames=eth5 eth6
sta_phy_ifnames=eth5 eth6
wan0_ifname=
wan0_pppoe_ifname=
wan1_ifname=
wan1_pppoe_ifname=
wan_ifname=
wan_ifnames=
wired_ifnames=eth1 eth2 eth3 eth4
wl0.1_ifname=wl0.1
wl0.2_ifname=wl0.2
wl0.3_ifname=wl0.3
wl0_ifname=eth5
wl0_vifnames=wl0.1 wl0.2 wl0.3
wl1.1_ifname=wl1.1
wl1.2_ifname=wl1.2
wl1.3_ifname=wl1.3
wl1_ifname=eth6
wl1_vifnames=wl1.1 wl1.2 wl1.3
wl_ifname=eth5
wl_ifnames=eth5 eth6
wl_vifnames=wl0.1 wl0.2 wl0.3

raoulteeuwen, 2021-03-03:
One question: this Asus is wired to & behind a primary router (the one offered by my ISP). Does that change how cake-qos works?

jackyaz, 2021-03-03:
Do you mean Asus is in AP mode?

raoulteeuwen, 2021-03-03:
> Do you mean Asus is in AP mode?

Yes, it currently is in AP mode (due to limitations enforced by my ISP, i would love to use the Asus as primary router). If that is the problem: excuse me from not mentioning that, and please consider adding the required mode to the list of Pre-requisites at https://github.com/ttgapers/cakeqos-merlin (maybe assumed obvious, sorry that it wasn't for me).

ttgapers, 2021-03-03:
> Yes, it currently is in AP mode (due to limitations enforced by my ISP, i would love to use the Asus as primary router). If that is the problem: excuse me from not mentioning that, and please [...]

Good point, and yes assumptions it's in router mode. I will add it to the list, and perhaps add a check if possible for mode the device is in as a pre-flight check. 

Will close this issue. Thanks for working through this.

closed 2021-04-28 by ttgapers