ttgapers/cakeqos-merlin #76: opkg missing is not reported by installer, which depends on it.

Yakumounr, 2021-06-25:
**Describe the bug**
Coming from a dsl-ax82u, with the 2 day old 386.02_6-gnuton1 which now has the kernel support for cake :

Neither the current v2.0.0. install script, nor the github cakeqos-merlin page or install instructions mention that entware is required.

The opkg failures that will be happening while running the script go unreported, the only installer hint you get that something may have failed on the ax82u is "ln: /opt/bin/cake-qos: No such file or directory" revealing a binary was not installed.

Though entware is easily installed via amtm (once you learn of amtm) it does also require a USB drive (excluding editing the installer to force it to jffs, which is discouraged), so the need for a USB drive should also be mentioned somewhere for cakeqos unless there are plans to remove the entware dependence, I assume cake doesn't benefit from a usb swap drive so the entware is the sole dependent on the drive?

**To Reproduce**
Follow the cakeqos-merlin install instructions without deviation reliant on prior knowledge from it's readme.md, on a clean install of 386.02_6-gnuton1, with no USB drive.

**Expected behavior**
opkg missing and being a mandatory requirement should be reported by the installer.
The documentation should also be amended to note it as an absolute requirement and what it is/how to get it, along with the requirement for a USB drive.

dave14305, 2021-06-25:
Where do you see a dependence on opkg? opkg is only called to remove the old Cake modules (if they are present) which would have been originally installed with opkg.

There is an potentially faulty assumption that the /opt/bin directory exists for creating a link to the script, but it’s not catastrophic if it fails to create it. No other dependency exists.

dave14305, 2021-06-25:
Make sure the gnuton build includes `cake` in the `rc_support` nvram variable. If not, you may be forced to download the old version of the script which does require Entware.

dave14305, 2021-06-25:
https://github.com/ttgapers/cakeqos-merlin/blob/386/README.md

Entware is documented in the legacy branches of this repo. It sounds like Gnuton hasn’t fully finished Cake support yet.

Yakumounr, 2021-06-25:
# nvram kget rc_support 
does show cake, as stated gnuton added the kernel support in 386.02_6-gnuton1 (stable), 
the UI doesn't work fully but manual input of 
# tc qdisc add dev br0 root cake
adds a cake rule to qdisc.

He had linked to the install script here with "[If you wanna test it...](https://github.com/gnuton/asuswrt-merlin.ng/issues/105#issuecomment-861347786)" noting that it still needed work, I was hoping it might be possible to get it working without too much difficulty.

opkg - you're right, just the two opkg --autoremove calls in v2.0.0 ,  it seems I'd tabbed back into https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/386/cake-qos.sh by mistake which also has the opkg install and list_installed that I remember. There was nothing more verbose than the missing /opt/bin/cake-qos message to give a lead, I hadn't looked deeper as a missing binary was a clear block, and I couldn't spend more time on it or try with entware installed until much later tonight.

My apologies, please nuke the thread :-)

dave14305, 2021-06-25:
Version 2.x of this script won’t work without the firmware generating the cake qos commands in /tmp/qos.

Yakumounr, 2021-06-26:
admin@AX82U:/tmp/home/root# /tmp/qos
```
class htb 1:10 parent 1:1 leaf 10: prio 1 quantum 1500 rate 10894Kbit ceil 15564Kbit linklayer ethernet burst 2959b/1 mpu 0b cburst 3542b/1 mpu 0b level 0
 Sent 20648206 bytes 285814 pkt (dropped 0, overlimits 0 requeues 0)
 rate 28424bit 42pps backlog 0b 0p requeues 0
 lended: 285814 borrowed: 0 giants: 0
 tokens: 32992 ctokens: 27785

class htb 1:1 root rate 15564Kbit ceil 15564Kbit linklayer ethernet burst 3542b/1 mpu 0b cburst 3542b/1 mpu 0b level 7
 Sent 124501067 bytes 892585 pkt (dropped 0, overlimits 0 requeues 0)
 rate 131440bit 205pps backlog 0b 0p requeues 0
 lended: 59880 borrowed: 0 giants: 0
 tokens: 27785 ctokens: 27785

class htb 1:2 root rate 10240Mbit ceil 10240Mbit linklayer ethernet burst 7680b/1 mpu 0b cburst 7680b/1 mpu 0b level 7
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 rate 0bit 0pps backlog 0b 0p requeues 0
 lended: 0 borrowed: 0 giants: 0
 tokens: 109 ctokens: 109

class htb 1:30 parent 1:1 leaf 30: prio 3 quantum 1500 rate 778Kbit ceil 15564Kbit linklayer ethernet burst 1696b/1 mpu 0b cburst 3542b/1 mpu 0b level 0
 Sent 22676083 bytes 33826 pkt (dropped 0, overlimits 0 requeues 0)
 rate 3528bit 1pps backlog 0b 0p requeues 0
 lended: 23835 borrowed: 9991 giants: 0
 tokens: 221708 ctokens: 25922

class htb 1:40 parent 1:1 leaf 40: prio 4 quantum 1500 rate 466Kbit ceil 15564Kbit linklayer ethernet burst 1657b/1 mpu 0b cburst 3542b/1 mpu 0b level 0
 Sent 80896397 bytes 572526 pkt (dropped 0, overlimits 0 requeues 0)
 rate 99456bit 162pps backlog 0b 0p requeues 0
 lended: 522726 borrowed: 49800 giants: 0
 tokens: 100281 ctokens: 27785

class htb 1:50 parent 1:1 leaf 50: prio 5 quantum 1500 rate 1867Kbit ceil 13229Kbit linklayer ethernet burst 1832b/1 mpu 0b cburst 3251b/1 mpu 0b level 0
 Sent 280381 bytes 419 pkt (dropped 0, overlimits 0 requeues 0)
 rate 32bit 0pps backlog 0b 0p requeues 0
 lended: 330 borrowed: 89 giants: 0
 tokens: 102967 ctokens: 27946

class htb 1:60 parent 1:2 leaf 60: prio 6 quantum 200000 rate 10240Mbit ceil 10240Mbit linklayer ethernet burst 7680b/1 mpu 0b cburst 7680b/1 mpu 0b level 0
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 rate 0bit 0pps backlog 0b 0p requeues 0
 lended: 0 borrowed: 0 giants: 0
 tokens: 109 ctokens: 109

class fq_codel 10:133 parent 10:
 (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  deficit 87 count 0 lastcount 0 ldelay 2us
class fq_codel 30:246 parent 30:
 (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  deficit -588 count 0 lastcount 0 ldelay 3us
class fq_codel 30:2af parent 30:
 (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  deficit -17 count 0 lastcount 0 ldelay 2us
class fq_codel 40:1d0 parent 40:
 (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  deficit 78 count 0 lastcount 0 ldelay 2us
class fq_codel 40:22e parent 40:
 (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  deficit -378 count 0 lastcount 0 ldelay 3us
class fq_codel 50:c parent 50:
 (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  deficit -280 count 0 lastcount 0 ldelay 4us
qdisc htb 1: root refcnt 2 r2q 10 default 0x40 direct_packets_stat 4 ver 3.17 direct_qlen 2
 Sent 124503403 bytes 892604 pkt (dropped 73, overlimits 27746 requeues 0)
[... 136 lines cut ...]
  deficit -60 count 0 lastcount 0 ldelay 3us
qdisc htb 2: root refcnt 2 r2q 10 default 0x40 direct_packets_stat 4 ver 3.17 direct_qlen 11000
 Sent 790304656 bytes 2369265 pkt (dropped 2403, overlimits 165866 requeues 0)
 backlog 0b 0p requeues 0
qdisc fq_codel 10: parent 2:10 limit 1000p flows 1024 quantum 300 target 5ms interval 100ms
 Sent 46518392 bytes 465765 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 1395 drop_overlimit 0 new_flow_count 6 ecn_mark 0
  new_flows_len 0 old_flows_len 1
qdisc fq_codel 30: parent 2:30 limit 1000p flows 1024 quantum 300 target 5ms interval 100ms
 Sent 16647303 bytes 35292 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 1500 drop_overlimit 0 new_flow_count 14941 ecn_mark 0
  new_flows_len 1 old_flows_len 16
qdisc fq_codel 40: parent 2:40 limit 1000p flows 1024 quantum 300 target 5ms interval 100ms
 Sent 726922328 bytes 1867510 pkt (dropped 2403, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 1504 drop_overlimit 0 new_flow_count 81567 ecn_mark 0
  new_flows_len 0 old_flows_len 1
qdisc fq_codel 50: parent 2:50 limit 1000p flows 1024 quantum 300 target 5ms interval 100ms
 Sent 216368 bytes 694 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 1500 drop_overlimit 0 new_flow_count 246 ecn_mark 0
  new_flows_len 0 old_flows_len 3
```

Yakumounr, 2021-06-26:
admin@AX82U:/tmp/home/root# /tmp/qos
full output:
[tmp-qos-outut.txt](https://github.com/ttgapers/cakeqos-merlin/files/6719319/tmp-qos-outut.txt)

Yakumounr, 2021-06-26:
If this is useful, or of any interest -
admin@AX82U:/tmp/home/root# /tmp/qos > [tqooutput.txt](https://github.com/ttgapers/cakeqos-merlin/files/6719330/tqooutput.txt)

closed 2021-07-05 by dave14305