ttgapers/cakeqos-merlin #79: Generalise to support different interfaces

lynxthecat, 2021-08-17:
Limiting to 'eth0' for WAN does not permit cake to be meaningfully applied in the context of the use of an OpenVPN because cake is operating blind on encapsulated packets designated for a single destination. 

This code should be generalised to support the application of cake to different interfaces in keeping with the OpenWRT SQM implementation, which permits selection of the appropriate interface and creates an IFB for the chosen interface, see:

https://github.com/tohojo/sqm-scripts/blob/master/src/layer_cake.qos

https://openwrt.org/docs/guide-user/network/traffic-shaping/sqm-details
 
That way the appropriate interface could be selected, e.g. 'tun11', and the appropriate WAN overhead set to take into account the VPN overhead (which can be easily determined using tcpdump and comparing packet sizes before and after encapsulation), see, for example, discussion here:

https://forum.openwrt.org/t/solved-sqm-with-a-vpn-what-interface-name-what-packet-overhead/57473

ttgapers, 2022-01-08:
Interesting post. I am unsure if this will be addressed in any future base release at this point. @lynxthecat have you attempted customizing the cake configs and test? Any interesting findings?

dave14305, 2022-01-08:
> Interesting post. I am unsure if this will be addressed in any future base release at this point. @lynxthecat have you attempted customizing the cake configs and test? Any interesting findings?

Lynx moved to OpenWrt and is doing some neat stuff there:
https://forum.openwrt.org/t/cake-w-adaptive-bandwidth/108848

closed 2022-01-08 by ttgapers

lynxthecat, 2022-01-10:
Well I still have an interest in AsusWrt Merlin - @dave14305 for example helped me a great deal when I used it. And I am keen to help with it where I can. 

@ttgapers yes worked fine in Asus Merlin. In OpenWrt it is totally common practice to apply SQM on different interfaces to deal with VPN, etc. 

I personally used this custom script:

```
#!/bin/sh

(
cat <<'ADDTEXT'
#!/bin/sh
source /etc/cake-qos.conf

TUN_PACKET_OVERHEAD=53

case "$1" in
start)
        if [[ -d /sys/class/net/tun11 ]]; then
                ULIF='tun11'
                OVERHEAD=$(echo $OVERHEAD | awk '$2=$2+'$TUN_PACKET_OVERHEAD)
        fi
                ip link add name ifb type ifb
                tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULOPTIONS $ULBW $OVERHEAD $FRAMING
                tc qdisc add dev $ULIF handle ffff: ingress
                tc qdisc add dev ifb root cake $DLPRIOQUEUE $DLOPTIONS $DLBW $OVERHEAD $FRAMING
                ip link set ifb up
                tc filter add dev $ULIF parent ffff: matchall action mirred egress redirect dev ifb
        ;;
stop)
        tc qdisc del dev eth0 ingress
        tc qdisc del dev eth0 root
        tc qdisc del dev tun11 ingress
        tc qdisc del dev tun11 root
        tc qdisc del dev ifb root
        ip link set ifb down
        ip link del ifb
        ;;
*)
esac
ADDTEXT
) > /tmp/qos
```

A challenge is dealing with mixture of VPN and non-VPN traffic. There is a solution using 'veth' and WireGuard - see here:

https://forum.openwrt.org/t/sqm-with-wireguard-and-pbr/105967

I think this same solution could be implemented in AsusWrt too, mindful that Wireguard support is just around the corner.

dave14305, 2022-01-10:
The last time I looked into it, veth wasn’t enabled in the Merlin kernel.

lynxthecat, 2022-01-10:
Where this is a will there is a way! Perhaps CAKE in the context of VPN is just not that much in demand? On my LTE router makes sense because I need VPN to circumvent traffic shaping by my ISP (throttling to 10 Mbit/s) and CAKE to prevent huge bufferbloat.