ttgapers/cakeqos-merlin #81: any statistics available

dtaht, 2021-11-24:
I was pleased to learn of this project today. Do your users have any visibility into any of the statistics cake keeps? I am primarily interested in backlog, drops, ack drops, and marks, but there are a variety of others.

underd0se, 2021-11-24:
@dtaht I'm a user of this project and yes, we have access to stats in the router GUI as well as to configuration and status information.  

[image]

ttgapers, 2021-11-25:
@dtaht thanks for  your comments/question. Appreciate it.

On the graphing etc., I believe this is something @dave14305 has been toying with for the future. @dave14305  would be better placed to comment on if/when/how that might be happening.

dtaht, 2021-11-25:
My overall campaign is to try and inform more users as to te value of dropping packets intelligently, and to do some a/b testing to pass the good word along. 

in your example above you might want to turn on the ack-filter on the upload. 

It would be good if the cake community could all agree on the right rules for re-marking videoconferencing traffic. 

I am also always curious as to what applications are using ecn, also. 

and seeing *live* stats, with a backlog, and happy users always is nice. thx for doing this port!

dave14305, 2021-11-26:
> @dtaht thanks for your comments/question. Appreciate it.
> 
> On the graphing etc., I believe this is something @dave14305 has been toying with for the future. @dave14305 would be better placed to comment on if/when/how that might be happening.

I’ve been using OpenWrt and digging into the multiple QoS/SQM efforts there (e.g. qosify, adapting QoS rules to nftables, etc.).

The big question was always, “What’s interesting in the CAKE stats output?” We have some feedback now from @dtaht to answer that question.

However I’m not sure if I’ll be switching back to Merlin firmware in the near future to work on this Addon.

dtaht, 2021-11-26:
Dream:

packets, ack-drops, drops, marks, on a log scale on one graph
backlog in bytes for each tin, linear scale on another graph
peak delay for each tin (also one graph), linear

there has been a variety of attempts to get there out there...

dave14305, 2021-11-26:
> packets, ack-drops, drops, marks, on a log scale on one graph
This is where it could be sloppy if we're dealing with 4 metrics for 1 to 8 tins (hopefully no one really uses diffserv8).

closed 2024-12-15 by ttgapers