ttgapers/cakeqos-merlin #87: Can't uninstall

gameinn1, 2023-02-06:
**Describe the bug**
I can't uninstall cake-qos

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh firmware reset
2. Do the install via mkdir -p /jffs/addons/cake-qos && /usr/sbin/curl -s "https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/master/cake-qos.sh" -o "/jffs/addons/cake-qos/cake-qos" && chmod 755 /jffs/addons/cake-qos/cake-qos && sh /jffs/addons/cake-qos/cake-qos install as instructed
3. Cake-Qos tab active and working
4. Run command cake-qos uninstall
5. Terminal states no such thing exists

**Expected behavior**
The tab would be removed

underd0se, 2023-02-06:
Connect to your router via SSH and type `cake-qos` to display an interactive menu which "uninstall" is the 4th option. Type 4 on that screen and hit enter.

gameinn1, 2023-02-06:
Ah thank you. If I could make a suggestion that the above post is put instead of https://github.com/ttgapers/cakeqos-merlin#uninstallremove

gameinn1, 2023-02-13:
I tried this, not working: https://i.imgur.com/QKN4mhX.png

sovanyio, 2024-12-17:
The script is under `/jffs/addons/cake...`.

I'm able to trigger the uninstall, but routing is broken when I do so, looks like there's something missing from the uninstall. Currently unable to workaround so I've reinstalled and routing is back to normal after this.

Even if QOS is disabled, seems like something is missing here.

ttgapers, 2025-01-30:
Hi,

As this code is pretty much frozen and I am unable to repro on my side. Note, I do not actively use this addon any longer as my internet connection no longer requires it at any of the site(s) I support. I would like to assist to get this issue closed however.

The uninstall function is shown below. Are you able to manually execute the steps to see if you end up in the same place? It would help in tracking down where there might be an issue.

`Cake_Uninstall(){
	printf "Removing WebUI...\n"
	prev_webui_page="$(sed -nE "s/^\{url\: \"(user[0-9]+\.asp)\"\, tabName\: \"${SCRIPT_NAME_FANCY}\"\}\,$/\1/p" /tmp/menuTree.js 2>/dev/null)"
	if [ -n "$prev_webui_page" ]; then
		# Remove page from the UI menu system
		LOCKFILE=/tmp/addonwebui.lock
		FD=386
		eval exec "$FD>$LOCKFILE"
		/usr/bin/flock -x "$FD"
		umount /www/require/modules/menuTree.js 2>/dev/null
		sed -i "\~tabName: \"${SCRIPT_NAME_FANCY}\"},~d" /tmp/menuTree.js
		if diff -q /tmp/menuTree.js /www/require/modules/menuTree.js >/dev/null 2>&1; then
			# no more custom pages mounted, so remove the file
			rm /tmp/menuTree.js
		else
			# Still some modifications from another script so remount
			mount -o bind /tmp/menuTree.js /www/require/modules/menuTree.js
		fi
		/usr/bin/flock -u "$FD"
		# Remove last mounted asp page
		rm -f /www/user/"$prev_webui_page" 2>/dev/null
	fi
	sed -i '\~# CakeQOS-Merlin~d' /jffs/scripts/service-event /jffs/scripts/service-event-end /jffs/scripts/services-start /jffs/scripts/firewall-start #/jffs/scripts/qos-start
	sed -i "/^cakeqos_/d" /jffs/addons/custom_settings.txt
	rm -rf "/opt/bin/${SCRIPT_NAME}" "${SCRIPT_DIR}" "/www/ext/${SCRIPT_NAME}" /jffs/configs/cake-qos.conf.add 2>/dev/null
}`

Please let me know. If you don't understand the code, I can summarize in a simpler step by step method.

closed 2025-01-30 by ttgapers