ttgapers/cakeqos-merlin #89: AC88U Support

KaMiKaZeE1221, 2024-06-30:
Any chance of getting support for the AC88U, I know it's only a dual core but I'm only running a 100/40 connection.
or if anyone knows how to install it on the AC88U please let me know :)

ttgapers, 2024-12-15:
Versions prior to 386.2 are supported on the legacy 386 branch or 384 branch. This script will automatically install the appropriate legacy version if your firmware does not support Cake natively.

closed 2024-12-15 by ttgapers

KaMiKaZeE1221, 2024-12-24:
> Versions prior to 386.2 are supported on the legacy 386 branch or 384 branch. This script will automatically install the appropriate legacy version if your firmware does not support Cake natively.

I only asked because I get this error, and that's on the 386 branch, but all good, thanks for replying anyway 🙂

[image]